ieeeupsb/Dora #21: Design CAD Model of Enclosure

G-Pereira, 2019-07-17:
Design an enclosure that fits a 100x100mm PCB. The fixings can be anywhere, just inform Electrical team where you plan to have screws or other types of fixings.

pedro-brands, 2019-07-17:
Portanto, eu vou fazer o design estrutural e o outro Pedro vai servir de consultor.

G-Pereira, 2019-07-18:
Ok, porreiro. nós vamos ter duas PCBs, uma mais junto ao chão com sensores de linha e outra principal. a ideia é fazer o robô o mais pequeno possivel e a PCB vai ter um máximo de 100x100mm (pode vir a ser mais pequena). Temos dois motores já no núcleo, as dimensões vão ser publicadas aqui #22 e por fim, temos de colocar vários LEDs (na verdade são sensores) a apontar para todas as direções (por volta de 12 LEDs) a uma altura razoável, por isso é preciso planear uma espécie de torre também. Tudo claro?

pedro-brands, 2019-07-18:
Muito bem, percebido! O luis já me tinha dado algumas indicações! Amanhã começo a fazer drafts de como pode ser o modelo e depois coloco aqui

G-Pereira, 2019-07-19:
Porreiro!