ieeeupsb/Dora #48: Track large files using LFS

morallito, 2021-03-09:
I'm see that Dora is back to life, but the repo is becoming big and big. In order to decrease the repository size to clone/push,  what do you think about tracking the large files using git lfs? 

Some references can be found [here.](https://git-lfs.github.com/)

morallito, 2021-03-09:
@fabiodrg is the GIT king, he can help you with it.

Sleepy105, 2021-03-10:
Hi. Yes, we are resurrecting this project.
The repo is getting big, but the individual project files are small. In the order of a few kilobytes. Git LFS is helpful when dealing with large files. The only exception is `electrical/core/fp-info-cache` with 2.9MB.

Sleepy105, 2021-03-10:
We should, however, add a proper .gitignore file to the project. This will reduce it's size and make sure we're only commiting useful information.

closed 2021-03-10 by Sleepy105